LibreCAD/LibreCAD #2279: Command Line and Aliases different behavior

Rehab42, 2025-08-29:
Version: 2.2.2.3-alpha<br/>Compiler: GNU GCC 13.3.0<br/>Compiled on: Aug 26 2025<br/>Qt Version: 6.4.2<br/>Boost Version: 1.83.0
System: Debian GNU/Linux 12 (bookworm)

I switched from Version: 2.2.0-alpha which was the .deb package in Gnomes Software Center to the one above to get the Dark Mode functional and when i close it doesn't spawn the open drawing in a new headless window. However my alias file has a custom Command: hl (horizontal) that i added that in the 2.2.0-alpha works but 2.2.2.3-alpha its not recognizing horizontal as a command line input. Was it removed from the command line?

sand1024, 2025-08-30:
@Rehab42 
well, as far as I can see, horizontal line command is still supported by command line. 

Default command is "linehor" or "lh"

Rehab42, 2025-08-31:
thank you @sand1024! seems the command changed name as prior in the 2.2.0-alpha it was just horizontal. Where is the documentation of the new command line aliases?