LibreCAD/LibreCAD #2280: UAF in RS_EntityContainer::optimizeContours()

jirislaby, 2025-08-30:
100% reproducible. If I _move and rotate_ the hatch inside the red triangle, I get a crash:
```
mouseEnterEvent(): resume: Move and Rotate Entities
mouseEnterEvent(): resume: Move and Rotate Entities
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_Hatch::update: invalid contour in hatch found
==1342666== Invalid read of size 8
==1342666==    at 0x2F19CE: RS_EntityContainer::optimizeContours() (rs_entitycontainer.cpp:1671)
==1342666==    by 0x302771: RS_Hatch::validate() (rs_hatch.cpp:129)
==1342666==    by 0x302B47: RS_Hatch::update() (rs_hatch.cpp:233)
==1342666==    by 0x4AB50E: UnknownInlinedFun (rs_modification.cpp:2099)
==1342666==    by 0x4AB50E: RS_ActionModifyMoveRotate::trigger() (rs_actionmodifymoverotate.cpp:66)
==1342666==    by 0x4A7A5F: RS_ActionModifyMoveRotate::mouseReleaseEvent(QMouseEvent*) (rs_actionmodifymoverotate.cpp:113)
==1342666==    by 0x398638: RS_EventHandler::mouseReleaseEvent(QMouseEvent*) (rs_eventhandler.cpp:177)
==1342666==    by 0x5116FB: QG_GraphicView::mouseReleaseEvent(QMouseEvent*) (qg_graphicview.cpp:543)
==1342666==    by 0x4B3CB57: QWidget::event(QEvent*) (qwidget.cpp:9045)
==1342666==    by 0x4AFB22D: QApplicationPrivate::notify_helper(QObject*, QEvent*) (qapplication.cpp:3640)
==1342666==    by 0x4B03499: QApplication::notify(QObject*, QEvent*) (qapplication.cpp:3084)
==1342666==    by 0x5AB54E7: QCoreApplication::notifyInternal2(QObject*, QEvent*) (in /usr/lib64/libQt5Core.so.5.15.17)
==1342666==    by 0x4B014ED: QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) (qapplication.cpp:2622)
==1342666==  Address 0x2100cc80 is 0 bytes inside a block of size 304 free'd
==1342666==    at 0x484C739: operator delete(void*, unsigned long) (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==1342666==    by 0x2EC62D: RS_EntityContainer::removeEntity(RS_Entity*) (rs_entitycontainer.cpp:480)
==1342666==    by 0x2F1AD5: RS_EntityContainer::optimizeContours() (rs_entitycontainer.cpp:1677)
==1342666==    by 0x302771: RS_Hatch::validate() (rs_hatch.cpp:129)
==1342666==    by 0x302B47: RS_Hatch::update() (rs_hatch.cpp:233)
==1342666==    by 0x4AB50E: UnknownInlinedFun (rs_modification.cpp:2099)
==1342666==    by 0x4AB50E: RS_ActionModifyMoveRotate::trigger() (rs_actionmodifymoverotate.cpp:66)
==1342666==    by 0x4A7A5F: RS_ActionModifyMoveRotate::mouseReleaseEvent(QMouseEvent*) (rs_actionmodifymoverotate.cpp:113)
==1342666==    by 0x398638: RS_EventHandler::mouseReleaseEvent(QMouseEvent*) (rs_eventhandler.cpp:177)
==1342666==    by 0x5116FB: QG_GraphicView::mouseReleaseEvent(QMouseEvent*) (qg_graphicview.cpp:543)
==1342666==    by 0x4B3CB57: QWidget::event(QEvent*) (qwidget.cpp:9045)
==1342666==    by 0x4AFB22D: QApplicationPrivate::notify_helper(QObject*, QEvent*) (qapplication.cpp:3640)
==1342666==    by 0x4B03499: QApplication::notify(QObject*, QEvent*) (qapplication.cpp:3084)
==1342666==  Block was alloc'd at
==1342666==    at 0x4848F3F: operator new(unsigned long) (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==1342666==    by 0x3127A2: RS_Line::clone() const (rs_line.cpp:79)
==1342666==    by 0x2F1CA9: RS_EntityContainer::optimizeContours() (rs_entitycontainer.cpp:1693)
==1342666==    by 0x302771: RS_Hatch::validate() (rs_hatch.cpp:129)
==1342666==    by 0x302B47: RS_Hatch::update() (rs_hatch.cpp:233)
==1342666==    by 0x4AB4A1: UnknownInlinedFun (rs_modification.cpp:2098)
==1342666==    by 0x4AB4A1: RS_ActionModifyMoveRotate::trigger() (rs_actionmodifymoverotate.cpp:66)
==1342666==    by 0x4A7A5F: RS_ActionModifyMoveRotate::mouseReleaseEvent(QMouseEvent*) (rs_actionmodifymoverotate.cpp:113)
==1342666==    by 0x398638: RS_EventHandler::mouseReleaseEvent(QMouseEvent*) (rs_eventhandler.cpp:177)
==1342666==    by 0x5116FB: QG_GraphicView::mouseReleaseEvent(QMouseEvent*) (qg_graphicview.cpp:543)
==1342666==    by 0x4B3CB57: QWidget::event(QEvent*) (qwidget.cpp:9045)
==1342666==    by 0x4AFB22D: QApplicationPrivate::notify_helper(QObject*, QEvent*) (qapplication.cpp:3640)
==1342666==    by 0x4B03499: QApplication::notify(QObject*, QEvent*) (qapplication.cpp:3084)
==1342666== 
==1342666== Invalid read of size 4
==1342666==    at 0x2F19E9: UnknownInlinedFun (rs_flags.cpp:50)
==1342666==    by 0x2F19E9: UnknownInlinedFun (rs_entity.cpp:197)
==1342666==    by 0x2F19E9: UnknownInlinedFun (rs_entity.cpp:195)
==1342666==    by 0x2F19E9: RS_EntityContainer::optimizeContours() (rs_entitycontainer.cpp:1671)
==1342666==    by 0x302771: RS_Hatch::validate() (rs_hatch.cpp:129)
==1342666==    by 0x302B47: RS_Hatch::update() (rs_hatch.cpp:233)
==1342666==    by 0x4AB50E: UnknownInlinedFun (rs_modification.cpp:2099)
==1342666==    by 0x4AB50E: RS_ActionModifyMoveRotate::trigger() (rs_actionmodifymoverotate.cpp:66)
==1342666==    by 0x4A7A5F: RS_ActionModifyMoveRotate::mouseReleaseEvent(QMouseEvent*) (rs_actionmodifymoverotate.cpp:113)
==1342666==    by 0x398638: RS_EventHandler::mouseReleaseEvent(QMouseEvent*) (rs_eventhandler.cpp:177)
==1342666==    by 0x5116FB: QG_GraphicView::mouseReleaseEvent(QMouseEvent*) (qg_graphicview.cpp:543)
==1342666==    by 0x4B3CB57: QWidget::event(QEvent*) (qwidget.cpp:9045)
==1342666==    by 0x4AFB22D: QApplicationPrivate::notify_helper(QObject*, QEvent*) (qapplication.cpp:3640)
==1342666==    by 0x4B03499: QApplication::notify(QObject*, QEvent*) (qapplication.cpp:3084)
==1342666==    by 0x5AB54E7: QCoreApplication::notifyInternal2(QObject*, QEvent*) (in /usr/lib64/libQt5Core.so.5.15.17)
==1342666==    by 0x4B014ED: QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) (qapplication.cpp:2622)
==1342666==  Address 0x2100cc88 is 8 bytes inside a block of size 304 free'd
==1342666==    at 0x484C739: operator delete(void*, unsigned long) (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==1342666==    by 0x2EC62D: RS_EntityContainer::removeEntity(RS_Entity*) (rs_entitycontainer.cpp:480)
==1342666==    by 0x2F1AD5: RS_EntityContainer::optimizeContours() (rs_entitycontainer.cpp:1677)
[... 36 lines cut ...]
==1342666==    by 0x2F19F6: RS_EntityContainer::optimizeContours() (rs_entitycontainer.cpp:1672)
==1342666==    by 0x302771: RS_Hatch::validate() (rs_hatch.cpp:129)
==1342666==    by 0x302B47: RS_Hatch::update() (rs_hatch.cpp:233)
==1342666==    by 0x4AB50E: UnknownInlinedFun (rs_modification.cpp:2099)
==1342666==    by 0x4AB50E: RS_ActionModifyMoveRotate::trigger() (rs_actionmodifymoverotate.cpp:66)
==1342666==    by 0x4A7A5F: RS_ActionModifyMoveRotate::mouseReleaseEvent(QMouseEvent*) (rs_actionmodifymoverotate.cpp:113)
==1342666== 
==1342666== HEAP SUMMARY:
==1342666==     in use at exit: 113,896,732 bytes in 298,499 blocks
==1342666==   total heap usage: 6,557,616 allocs, 6,259,117 frees, 1,052,424,855 bytes allocated
==1342666== 
==1342666== LEAK SUMMARY:
==1342666==    definitely lost: 384,048 bytes in 1,417 blocks
==1342666==    indirectly lost: 1,000,468 bytes in 1,031 blocks
==1342666==      possibly lost: 1,420,526 bytes in 10,128 blocks
==1342666==    still reachable: 111,089,790 bytes in 285,900 blocks
==1342666==                       of which reachable via heuristic:
==1342666==                         newarray           : 624 bytes in 16 blocks
==1342666==                         multipleinheritance: 350,512 bytes in 419 blocks
==1342666==         suppressed: 52 bytes in 2 blocks
==1342666== Rerun with --leak-check=full to see details of leaked memory
==1342666== 
```

[drawing.tar.gz](https://github.com/user-attachments/files/22058793/drawing.tar.gz)

sand1024, 2025-08-30:
@jirislaby 
Could please also mention your version of librecad?

dxli, 2025-08-30:
@jirislaby 

it would also help if you upload a dxf file (in a zipball) showing this issue.

Please test the dxf with the latest development version.

In any case, the hatching algorithms will see some changes to improve robustness and correctness.

jirislaby, 2025-08-31:
> Could please also mention your version of librecad?

2.2.1.2

> it would also help if you upload a dxf file (in a zipball) showing this issue.

I already did -- Is it not visible in the first post?

> Please test the dxf with the latest development version.

OK, I will.

jirislaby, 2025-09-01:
> I can open the dxf. It's a complex drawing, so I'm not sure on how to reproduce this issue. Could you elaborate on the the problematic hatch? For the latest development version.

OK, I removed basically everything from the drawing except the problematic hatches inside the triangles. Whatever you try to do with the hatches, it's done wrong. They are somehow special -- note that selecting one hatch, selects all of them. Rotating or moving them distorts them.

[drawing.tar.gz](https://github.com/user-attachments/files/22073090/drawing.tar.gz)