LibreCAD/LibreCAD #2309: Setting to revert context menu back to only showing recently used tools

Questions2000, 2025-09-29:
I only very quickly tested out this newest version, I apologize if there is already a setting for this.

I noticed that the drawing pane RMB context menu now contains tool flyouts and the list or recently used tool is also now in a flyout called `Recent`.

In certain situations these tool flyouts can be helpful to some users, but this can make the context menu a bit larger then desired.  This also makes the recent tool list require a bit more effort to access now that it is in a flyout itself, which is not as convenient as before to quickly reapply an action.

If there is not setting for this, I would consider adding option to revert back to the previous CM state that only shows recent actions performed.

Thank You

### Operating System and LibreCAD Version
Version: LibreCAD-2.2.2_alpha1-473-g606035f42.AppImage
Compiled on: Sept 18 2025
System: Debian GNU/Linux 12 (bookworm) / GTK based

sand1024, 2025-09-30:
@Questions2000
Well, adding an option for this may be a good idea indeed. It will be up to the user to decide, whether it's menu fits to the screen or not. 

yet in general, the entire context popup menus were reworked - and now it's actually the "context" one. 
You can find more details about changes there - https://github.com/LibreCAD/LibreCAD/pull/2286

Questions2000, 2025-10-05:
Thank you guys for the replies.
I have some ideas for this but would take some time to type them up and hard to know if they will be accepted.